riscv-non-isa/riscv-sbi-doc #2: getchar/putchar interface

sam-falvo, 2018-08-07:
I notice that these functions take or result in an `int`.  Are these interfaces intended to take only 7-bit ASCII, or should they be Unicode codepoints?

The reason I'm asking is because of `getchar`'s use of -1 to indicate failure to retrieve a character.  With a wrapper around `getchar`, you can use this to implement a function `can_getchar`, which returns true if and only if there's a character waiting, but which does not pop the character input queue.  However, doing so involves locally caching the retrieved character, which seems like it should be the job of whatever is implementing the SBI.

Maybe three functions are to be preferred?

* int getchar() -- retrieves the next console character.  Never blocks; -1 if not possible to grab character.
* int can_getchar() -- true if and only if getchar() will yield valid data.
* putchar(int) -- sends a character out to the console.

Just an idea.

sam-falvo, 2018-08-07:
Never mind; I missed that they are spec'd specifically to send or receive *bytes*, which implies ASCII only operation.  Use of -1 as a sentinel is safe then.

closed 2018-08-07 by sam-falvo

neuschaefer, 2018-08-07:
Byte-based non-ASCII encodings such as UTF-8 should also work.

sam-falvo, 2018-08-07:
Mnemonically, then, I'd recommend changing the names of the functions to `sbi_console_{get|put}byte` instead of `sbi_console_{get|put}char`, to make the distinction between characters and their encodings more clear.