riscv-non-isa/riscv-sbi-doc #4: Return values

neuschaefer, 2018-09-07:
Hi, currently one SBI call (`sbi_console_getchar`) returns a value, but the document doesn't explicitly specify *how* the value is returned (through register a0).

I am specifically interested in a corner case: If an SBI call is specified with the `void` return type, Is an SBI implementation allowed to set a0 to a dummy value, such as 0?

aswaterman, 2018-09-07:
All SBI calls should be assumed to clobber a0. This should be reflected in the spec.

atishp04, 2018-09-14:
Thanks. I will update the doc

palmer-dabbelt, 2018-09-19:
IIRC just documenting this as being the same as the system call interface (which aside from a7 the same as the function call interface) should cover all these sorts of edge cases.  The intent is that they're all the same to avoid confusion.

That's already documented in the spec, so I'm going to close this.  Feel free to re-open it if my assumption is not actually the case.

closed 2018-09-19 by palmer-dabbelt