riscv-non-isa/riscv-sbi-doc #8: Function numbers are not specified

programmerjake, 2018-10-12:
The numbers to put in `a7` are not specified.

atishp04, 2018-10-12:
#define SBI_SET_TIMER 0
#define SBI_CONSOLE_PUTCHAR 1
#define SBI_CONSOLE_GETCHAR 2
#define SBI_CLEAR_IPI 3
#define SBI_SEND_IPI 4
#define SBI_REMOTE_FENCE_I 5
#define SBI_REMOTE_SFENCE_VMA 6
#define SBI_REMOTE_SFENCE_VMA_ASID 7
#define SBI_SHUTDOWN 8

I will add it to the function table.

atishp04, 2018-10-17:
Updated. 

@aswaterman : I was experimenting something and got the commit directly merged to master of riscv/riscv-sbi-doc instead of my branch(I did not know I have write access :)). 

Here is the commit (minor change just to update the function IDs).
https://github.com/riscv/riscv-sbi-doc/commit/877a695b879e7a4836afffefc26c29c7bdadd49e

Let me know if I should revert this and follow the standard process of sending a PR.

palmer-dabbelt, 2018-10-18:
I think it's fine.

atishp04, 2018-10-18:
Thanks. Closing the issue.

closed 2018-10-18 by atishp04