riscv-non-isa/riscv-sbi-doc #17: Does sbi_send_ipi really take a physical address?

fintelia, 2019-05-23:
The current document says that the argument to sbi_send_ipi should be a physical address. However, all existing implementations I could fine actually expect virtual rather than physical addresses:

* BBL's [send_ipi_many](https://github.com/riscv/riscv-pk/blob/cb253a3f970b373df4390a0e7a17610556f920f3/machine/mtrap.c#L102) / [load_##type](https://github.com/riscv/riscv-pk/blob/cb253a3f970b373df4390a0e7a17610556f920f3/machine/unprivileged_memory.h#L11) uses the MPRV mechanism to load a supervisor virtual address (and I can confirm from testing that it page faults if the argument is only valid as a physical address). 
* OpenSBI's code is almost exactly the same: [sbi_ipi_send_many](https://github.com/riscv/opensbi/blob/4e2cd478208531c47343290f15b577d40c82649c/lib/sbi_ipi.c#L154) / [load##type](https://github.com/riscv/opensbi/blob/10baa64c02f6746fd506136e0693aa2d592574fb/include/sbi/riscv_unpriv.h#L18) 
* [Linux also passes a virtual address to sbi_send_ipi](https://github.com/torvalds/linux/blob/39feaa3ff4453594297574e116a55bd6d5371f37/arch/riscv/kernel/smp.c#L146).

atishp04, 2019-06-03:
You are correct. Physical address was a typo. Sent a PR to fix it. #18 

Thanks for pointing it out.

palmer-dabbelt, 2019-06-13:
I'm closing this, as I believe the issue has been resolved by #18.  Thanks!

closed 2019-06-13 by palmer-dabbelt