riscv-non-isa/riscv-sbi-doc #25: Why does SBI have no way to get a hart id?

sux2mfgj, 2019-09-28:
There is no way to get a hart id (csr mhartid) via SBI. Is there another way or any reason?
Looks linke the current RISC-V Linux implementation refers to device tree.
https://github.com/torvalds/linux/blob/master/arch/riscv/kernel/cpu.c#L25

xy-plus, 2019-10-12:
I meet the same problem with you while writing a toy os.

I get the hart id by this way:

1. use asm rewrite _start

In _start, I set the stack for my_main, then I call my_main.

2. my_main first arg is the hart id, second arg is device tree base. It means R[a0] is hart is and R[a1] is device tree base.

here is my code, hope it can help you.

```nasm
    .section .text.entry
    .globl _start
_start:
    lui sp, %hi(bootstacktop)

    call rust_main

    .section .bss.stack
    .align 12  # PGSHIFT
    .global bootstack
bootstack:
    .space 4096 * 4
    .global bootstacktop
bootstacktop:
```

```rust
#[no_mangle]
pub fn rust_main(hartid: usize, dtb: usize) -> ! {
    loop {}
}
```

aswaterman, 2020-03-04:
Because there’s no need for it, strictly speaking. Being passed the hart id at boot is sufficient. It can be stored in a thread-local structure. Read the RISC-V Linux kernel source code for one example of a system that works this way.

atishp04, 2020-03-04:
@JohnAZoidberg As Andrew pointed out, a0 register can be used to pass the hartid.
If you are concerned about EDK2 implementation, there is already ongoing discussion and patch to add a DT entry for the boot hartid that EFI stub will parse and set a0 before jumping to Linux kernel

https://lists.denx.de/pipermail/u-boot/2020-February/401562.html

atishp04, 2020-10-19:
closing this issue as it is resloved.

closed 2020-10-19 by atishp04

agrobman, 2021-11-04:
@aswaterman , 

Serving  external interrupts  in multicore system, I need to know what PLIC addresses to access. M-mode handler can use mhartid to calculate the PLIC registers address or any hart related data structures.
However S-mode handler does not have access to mhartid. 
How would such handler get pointer to the hart-local data structure, if all harts execute the same handler code?

avpatel, 2021-11-04:
The hartid will be passed in a0 register for each HART so Software can save this information in some per-hart data structure.

Also, the sscratch CSR can be used to point to per-HART data structure which holds hartid information.

Regards,
Anup