riscv-non-isa/riscv-sbi-doc #26: Add reset function

aurel32, 2019-10-06:
The SBI specification provides a shutdown function, but no reset function. Reset support has been added recently to the QEMU "RISC-V VirtIO Board" machine. Would it be possible to add such a function to the SBI specifications? It's not clear to me what process has to be followed to update the specifications.

atishp04, 2019-10-06:
We are in the process of sending a patch to add the reset function. I will
send the PR in couple of days.

atishp04, 2019-10-16:
@aurel32 : Reset function added
https://github.com/riscv/riscv-sbi-doc/pull/27

aurel32, 2019-10-17:
@atishp04 Thanks a lot, this looks all good (I only checked the shutdown/reboot part).

shlevy, 2020-03-17:
Looks like this got removed before that PR was merged?

avpatel, 2020-03-18:
We had intentionally kept reset extension out of SBI v0.2.

Now that SBI v0.2 is tagged and ready to be ratified, I will send SBI reset extension patch again for review.

Regards,
Anup

atishp04, 2020-10-19:
The reset extension is under discussion and there is a PR available #39 . Closing this issue.

closed 2020-10-19 by atishp04