riscv-non-isa/riscv-sbi-doc #33: v0.2: Timer and IPI extensions are missing "Function Listing" tables

konrad-schwarz, 2020-03-04:
The Timer and IPI extensions are missing the "Function Listing" tables present in the other extensions.

Although the Timer and IPI extensions have only a single function each, it would be
forward-thinking to define the appropriate function IDs to enable extension in the future.

Also, without function IDs, the switch from the v0.1 legacy encoding to the standard encoding provides no benefit.

atishp04, 2020-03-04:
I agree. Function IDs was assumed to be zero as the it is the only function. But it should be documented in the specification. I will send a patch. Thanks.

atishp04, 2020-10-19:
The requested change is already merged. Closing this issue.

closed 2020-10-19 by atishp04