riscv-non-isa/riscv-sbi-doc #35: Status of register x1 (ra) in SBI ABI calling convention

konrad-schwarz, 2020-03-06:
Is x1 -- the standard return address register -- callee or caller saved in the SBI ABI?  ECALL obviously does not require it, storing the return address in the epc CSR.

From a technical standpoint, I guess it would make slightly more sense to make it callee saved, as SBI calls that do not require ra, because they don't make further subroutine calls or only call "milli-code" routines using the alternative t0 register, would run slightly faster.

luojia65, 2021-11-22:
If we regard `ecall` as the jump instruction to call a SBI function, we may consider that the caller only fill parameters into only `a0` to `a7` registers (without touching the stack), and read return values from only `a0` and `a1` registers. Then, all the remaining registers except for `x0` should be saved by callee to not breaking the SBI function call convention (if I am correct).

aswaterman, 2021-11-22:
The Linux kernel assumes that all registers besides `a0` and `a1` are preserved across the SBI call: https://github.com/torvalds/linux/blob/136057256686de39cc3a07c2e39ef6bc43003ff6/arch/riscv/kernel/sbi.c#L38-L41

konrad-schwarz, 2021-11-23:
Shouldn't this be specified in the SBI spec? 

Also, I find it surprising that the temporary registers are preserved -- surely this is suboptimal?  I.e., an environment call is very similar to a function call; the reasoning behind the split between caller-saved and callee-saved registers must be largely the same.

cmuellner, 2021-11-23:
The calling convention is documented in the SBI specification: https://github.com/riscv-non-isa/riscv-sbi-doc/blob/master/riscv-sbi.adoc#binary-encoding

```
[...] This binary encoding matches the RISC-V Linux syscall ABI, which itself is based
on the calling convention defined in the RISC-V ELF psABI.
In other words, SBI calls are exactly the same as standard RISC-V function calls except
that [...] a7 (or t0 on RV32E-based systems) encodes the SBI extension ID.
```

The [RISC-V CC](https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/master/riscv-cc.adoc) states, that `x1` is not preserved across calls.

Based on this specified behavior, `x1` needs to be caller-saved.

If the implementation differs from the specification, there is a bug.

The code in the Linux kernel (`sbi_ecall`) works because the function is not inlined (so the caller of `sbi_ecall` saves `x1`).

avpatel, 2021-11-23:
(My response from platform mailing list)

The SBI calling convention only defines uses of a0-a7 registers for
parameter passing and return value because all other registers are
assumed to be saved/restored (i.e. preserved) by the SBI
implementation (M-mode firmware or Hypervisor). This is based on the
following rationale:

1) The SBI caller could be really anything. It could be some
bare-metal application with it's own register calling convention or it
could be some proprietary OS with proprietary ABI. This means SBI
implementation cannot make any assumptions about the callers (i.e. SBI
clients).

2) Most SBI implementations (OpenSBI, BBL, etc) are also responsible
for emulating missing HW functionality (such as emulating misaligned
load/store, time CSR, htimedelta CSR, CMO instructions) and the
trapping instructions could be touching any register (tX or aX or sX).
Further in RISC-V, we have the same mtvec CSR for handling SBI calls,
interrupts, misaligned load/store, and illegal instructions so SBI
implementations need trap handlers to save all registers anyway. This
means SBI calling convention has to be aligned with this aspect of
RISC-V and ensure that all registers are preserved by SBI
implementations.

Some aspects of point#2 above also apply to Linux kernel and that's
why we have handle_exception() of <linux>/arch/riscv/kernel/entry.S
preserving all registers.

Regarding performance of SBI calls, we have following options at the moment:

1) It is very well known that SBI IPI, Timer, and RFENCE calls can
easily be in the hot-path of many use cases/workloads. The upcoming
Sstc, AIA, and ACLINT specifications allow RISC-V platforms to totally
avoid SBI IPI, Timer, and RFENCE calls so it is upto RISC-V platforms
to decide how much they want to depend on SBI calls for fundamental HW
functionality like IPI, Timer, Rfence, etc.

2) In future, we might have standard instructions to load/store
multiple registers and SBI implementations can use such instructions
to optimize their low-level/assembly trap handlers.

Based on the above points, I think the SBI calling convention does the
right thing by only touching a0-a7 registers forcing SBI
implementations to preserve the rest of the registers.

konrad-schwarz, 2021-11-23:
> The Linux kernel assumes that all registers besides a0 and a1 are preserved across the SBI call
!=
> The code in the Linux kernel (sbi_ecall) works because the function is not inlined (so the caller of sbi_ecall saves x1).

And honestly, this reasoning is bogus.  How do you know GCC is not using a temporary register in sbi_ecall, or will not do so in some future revision of the code generator?  Why not explicitly state that the temporary registers can be clobbered in the inline __asm__ statement?

avpatel, 2021-11-23:
> The calling convention is documented in the SBI specification: https://github.com/riscv-non-isa/riscv-sbi-doc/blob/master/riscv-sbi.adoc#binary-encoding
>
> ```
> [...] This binary encoding matches the [...]

You are assuming that caller is following psABI but it could some bare-metal application or some proprietary OS with proprietary ABI. In other words, the SBI caller could be using caller saved register as callee saved.

Also, if we don't preserve all registers then misaligned load/store, illegal instruction emulation, and interrupt handling goes for a toss in the SBI implementation.

Regards,
Anup

cmuellner, 2021-11-23:
@konrad-schwarz: I assume you misunderstood my comment. I did not defend the kernel implementation or even claim it correct. I just stated why it is working. I fully agree with you, that the ASM statement should be brought in line with the specification.

avpatel, 2021-11-23:
I think SBI spec does the right thing by not saying anything about preserving other registers and leaving it to the SBI implementations.

This means SBI implementations have two options:

1) SBI implementations (such as OpenSBI, KVM, Xvisor, etc) which want to support wide range of SBI clients/callers (including proprietary and bare-metal applications) can do so by preserving all registers. This will also help them do misaligned load/store emulation, CSR emulation, and interrupt handling.

2) Very specific SBI implementations which assume specific ABI followed by SBI callers/clients and can optimise register save/restore accordingly. This obviously has downside not being able to support any kind of SBI callers/clients.

If we still need clarification in the SBI spec then a non-normative text would be suitable.

Regards,
Anup

konrad-schwarz, 2021-11-23:
> I think SBI spec does the right thing by not saying anything about preserving other registers and leaving it to the SBI implementations.

I'm sorry, but this is nonsense.  SBI implementations and SBI users need to conform to the SBI spec to be able to interoperate -- this the whole purpose of a **binary interface**: integration of independently developed modules.  What they do internally is of no interest at all -- if they are bare metal, have proprietary function linkage conventions or anything else you have mentioned.

As such, the SBI *must* define how the temporary registers are to be handled.  For reasons described above, and for backwards compatibility ("SBI calls are exactly the same as standard RISC-V function calls"), temporary registers should be defined to be caller-saved.  Since there is obviously confusion about what the SBI spec requires (perhaps, misguidedly, on purpose), it would be good to clarify the wording.

avpatel, 2021-11-23:
Like mentioned previously, we can't assume ABI followed by SBI caller/client so currently almost all SBI implementations (OpenSBI, BBL, KVM, Xvisor, Bao, etc) preserve all registers.

SBI implementations can't enforce partial preservation of registers for interoperability with proprietary and bare-metal applications.

Regards,
Anup

atishp04, 2021-11-23:
> I think SBI spec does the right thing by not saying anything about
> preserving other registers and leaving it to the SBI implementations.
>
> I'm sorry, but this is nonsense. SBI implementations and SBI [...]
If interoperability is the sole concern, it should be added to the platform
spec which ensures the interoperability. IMO, it should be the
implementation's responsibility based on their type of client expectation.
We shouldn't burden the callers with save/restore who don't care about
interoperability.

konrad-schwarz, 2021-11-23:
> Like mentioned previously, we can't assume ABI followed by SBI caller/client so currently almost all SBI implementations (OpenSBI, BBL, KVM, Xvisor, Bao, etc) preserve all registers.

As authors of the SBI Specification, you don't have to *assume* anything.  The SBI specification must *prescribe* what is to be done.
 
> SBI implementations can't enforce partial preservation of registers for interoperability with proprietary and bare-metal applications.

"Interoperability with proprietary and bare-metal applications" is a red herring.  If the SBI prescribes temporary registers to be caller-saved, then all applications, proprietary, bare-metal or not, must implement their end of the bargain.

> If interoperability is the sole concern, it should be added to the platform spec which ensures the interoperability.

Why invoke the platform spec?   This matter can and should be fully handled within the SBI spec itself,
the definition of callee-saved registers clearly being in scope of an ABI specification.

> IMO, it should be the implementation's responsibility based on their type of client expectation. We shouldn't burden the callers with save/restore who don't care about interoperability.

In reality you have achieved the worst of all possible outcomes, since according to your reading neither the SBI implementation nor the SBI user can assume anything about temporary registers, burdening both *both* with the requirement to save/restore them.

cmuellner, 2021-11-23:
@atishp04: If one does not care about a calling convention, why would that person then care about anything in the SBI specification (e.g. EIDs)?

Additionally, if you specify registers to be preserved, then you don't need to save/restore them in your SBI implementation. You can also choose to simply not use the registers. But that is an implementation detail.

For sure the sentence `In other words, SBI calls are exactly the same as standard RISC-V function calls except
that [...] a7 (or t0 on RV32E-based systems) encodes the SBI extension ID.` is misleading if the psABI CC is not applicable.

Proposal:
Would it be possible to define an SBI-CC, that can be referred to, if applicable (i.e. if interoperability is relevant)?
If that is kept in a separate document, then it is possible for users to choose what to implement.

If you insist on strictly not defining any SBI-CC (for whatever reason), then would it be possible to explicitly state, that no assumptions regarding unmentioned register preservation can be made and thus all these registers have to be considered caller-saved unless the actual implementation (e.g. OpenSBI) provides more guarantees. An information suggestion to provide the actual implemented SBI-CC along with the documentation of an SBI implementation might be a good idea. Btw, does anyone know if OpenSBI has its register preservation guarantees documented?

atishp04, 2021-11-23:
> @atishp04 <https://github.com/atishp04>: If one does not care about a
> calling convention, why would that person then care about anything in the
> SBI specification (e.g. EIDs)?

Maybe you misunderstood my statement. I meant to say that one may not care
about saving all the registers if they don't care about interoperability.

> Additionally, if you specify registers to be preserved, then you don't
> need to save/restore them in your SBI implementation. You can also choose
> to simply not use the registers. But that is an [...]
On the contrary, psABI CC is applicable for everything that must follow
psABI spec. So we don't need to define a separate SBI CC for that.
For the bare metal & proprietary applications which don't need to follow
psABI, we can leave it to the implementation to do the right thing as per
their requirement.

> Proposal:
> Would it be possible to define an SBI-CC, that can be referred to, if
> applicable (i.e. if interoperability is relevant)?
> If that is kept in a separate document, then it is possible for users to
> choose what to implement.

Having said that, if the consensus is to define a separate SBI-CC, do we
need to define it for both the use cases defined above ?
or just mandating the caller save approach for everything?

> If you insist on strictly not defining any SBI-CC (for whatever reason),
> then would it be possible to explicitly state, that no assumptions
> regarding unmentioned register preservation can be made [...]

cmuellner, 2021-11-23:
Regarding psABI: we have the CC, the DWARF register mapping and the ELF details defined there. But it is perfectly fine to ignore sections if not applicable.

The idea is, that we want to avoid pointless fragmentation. People who want to come up with their own specifications cannot be stopped from doing so. But everyone who wants to be interoperable should have a clear specification to work with. I would argue that many implementors want an interoperable solution. However, they don't have a specification as of now (that's what I understood from the responses here).

I'm not sure if I understood your last paragraph.
If the question was regarding how the solution should look like, then I think that's something somebody should work out a proposal, present it, collect feedback (Linux, FreeBSD, embedded solutions), and finalize it.

avpatel, 2021-11-24:
If we are to define a SBI-CC then all SBI implementations should preserve both caller-saved and callee-saved registers because most SBI implementations (OpenSBI, BBL, KVM, Xvisor, etc) will anyway end-up saving all registers in order to support misaligned load/store emulation, CSR emulation, etc. This means SBI clients should not be required to save anything.

If we SBI-CC defines that caller-saved need to be preserved by SBI clients then still most SBI implementations will have to preserve all  registers for supporting instruction and CSR emulation. Additionally, the SBI caller saving caller-saved registers will be end-up being totally redundant.

Regards,
Anup

konrad-schwarz, 2021-11-24:
I'm sorry, but to be blunt none of this makes any sense!

> If we are to define a SBI-CC then all SBI implementations should preserve both caller-saved and callee-saved registers because most SBI implementations (OpenSBI, BBL, KVM, Xvisor, etc) will anyway [...]

The *current* wording makes it fairly clear that SBI clients must save temporary
registers (i.e., the `t0`--`t6` registers defined as callee-saved by the standard
RISC-V ABI).  This is also technically the right thing to do, for exactly the same reasons
why the RISC-V ABI has defined them as callee saved in the first place.

"Mis-aligend load/store emulation, CSR emulation, etc." have absolutely **NO** bearing on the question at hand and I fail to see why this is being brought up.
 
> If we SBI-CC defines that caller-saved need to be preserved by SBI clients then still most SBI implementations will have to preserve all registers for supporting instruction and CSR emulation. Additionally, the SBI caller saving caller-saved registers will be end-up being totally redundant.

"most SBI implementations will have to preserve all registers for supporting instruction
and CSR emulation"  -- this is not true.  Although on RISC-V, exception
and interrupt handling are (in lieu of vectored interrupts) channeled through
the same base vector, register save code (i.e., context save) does not have to be
concentrated at the start of this vector.  An implementation
can defer register save/restore until it is known what is needed, leading
to faster exception response times.

If a specific implementation, e.g. OpenSBI, chooses to  save all registers
on every exception entry, fine -- this would not
violate the standard -- but that implementation's choice should not
preclude improved solutions.

> For SBI implementations, there is really no choice to partially save registers because there are additional responsibilities (instruction and CSR emulation) along with supporting SBI calls.

There definitely is a choice to partially save registers: as outlined above,
by saving only a minimal subset of registers at the start of the exception vector
and by adding more registers as necessary during the processing of the
specific exception cause.

aswaterman, 2021-11-24:
Like it or not, we have a de facto convention: Linux assumes registers other than a0/a1 are preserved. There’s no point in arguing about it. We should document it and be done with the philosophizing on the matter.

konrad-schwarz, 2021-11-24:
What precludes Linux from receiving the one-line patch to ensure it is not relying on the SBI to preserve the t0--t6 registers -- which by accident of implementation it is already not doing, as the SBI function is not in-lined (or at least the functions calling it are not) and itself does not require any preservation of the temporary registers over the ECALL instruction, and leaving the OpenSBI implementation as it is for the time being?

This would cause Linux to become conforming with the existing SBI specification and is the right thing to do from a technical/efficiency standpoint.

I'll also note that the ECALL-related code in sbi.c is unnecessarily bloated, writing to all 8 argument registers on every invocation and precluding generation of li instructions for the constants in a7 and a8.  A more efficient solution is readily achieved through use of the GNU macro assembler; I can provide code if requested.

aswaterman, 2021-11-24:
IMO, incompatibility between newer SBI implementations and existing Linux kernels precludes that.  Better to fix the spec than to create this software incompatibility.

jrtc27, 2022-01-12:
> Like it or not, we have a calling convention: Linux assumes registers other than a0/a1 are preserved. There’s no point in arguing about it. We should document this interface and dispense with the philosophizing on the matter.

FreeBSD has also always assumed this, because that's what BBL did.

jrtc27, 2022-01-12:
Well, in fact, it used to assume a1 was also preserved, because that's what BBL did and it didn't participate in the calling convention. Since adding support for non-legacy SBI base and extensions it also marks a1 as an output, and thus clobbered. But it is crucial that any legacy function only clobber a0, otherwise you break backwards compatibility with software written against BBL. Base and other SBI extensions are free to clobber a1 as well.

atishp04, 2022-01-27:
@palmer-dabbelt sent a fix for Linux kernel. It will be back ported to older versions as well.

https://lkml.org/lkml/2022/1/27/1147

With this, we don't require any changes to the SBI spec which just says that we should follow the standard calling convention defined by psABI.

darius-bluespec, 2022-01-27:
> With this, we don't require any changes to the SBI spec which just
  says that we should follow the standard calling convention defined
  by psABI.

It also says the "binary encoding matches the RISC-V Linux syscall
ABI", which I believe treats all registers not used as arguments or
return values as callee saved.

So which is it?

aswaterman, 2022-01-27:
@palmer-dabbelt's patch should be reverted and the spec should be fixed to resolve the contradiction @darius-bluespec pointed out in favor of what the software actually does (i.e., the calling convention matches the Linux syscall ABI).

All the software was working just fine before, and all we had to do is fix this contradiction.  I don't understand why we think it's reasonable to do the more painful thing that also reduces performance.

jrtc27, 2022-01-27:
Agreed. Were we to design this from scratch we would probably follow the normal psABI calling convention (which is what Arm do for their firmware call interface), but we're not doing that, we have years of existing OS releases that we cannot break.

palmer-dabbelt, 2022-01-27:
Sorry, I hadn't checked that syscalls matched the psABI (I'd just assemed so because that line makes it look like so).  I don't really care all that much which way this goes, but the spec needs to at least be internally consistent.

atishp04, 2022-01-31:
Thanks @darius-bluespec pointing out about the syscall ABI. I think its better to fix the SBI spec as per @aswaterman's suggestion with the following line. The actual patch will remove the statement about syscall ABI & psABI.

"All registers except a0 & a1 must be preserved across SBI call."

atishp04, 2022-01-31:
> > "All registers except a0 & a1 must be preserved across SBI calls"
> 
> Aren't you forgetting a7 and a6, used to hold the SBI Extension ID (EID) and Function ID (FID) respectively? And what about SBI calls that would benefit from more than two (true) arguments?
> 

The existing content that specifies about the a7/a6 will continue to be there. I just specifying about the registers that don't need to be preserved across the SBI call here. As andrew mentioned earlier, Linux clobbers expects all registers except a0 & a1 to be preserved. Hence the above suggestion.

> I really think its a shame you can't be convinced to stay as close to the standard (SysV ELF) ABI calling conventions as possible, (i.e., extended with the EID/FID use of a6/a7 and use of ecall instead of jalr). A lot of thought went into the ABI: the SBI spec would benefit from that, it would be easy to remember for assembly programmers, and e.g., commercial or experimental compilers lacking GCC's excellent inline asm support would have an easier time of it.

konrad-schwarz, 2022-02-01:
>>> "All registers except a0 & a1 must be preserved across SBI calls"

>> Aren't you forgetting a7 and a6, used to hold the SBI Extension ID (EID) and Function ID (FID) respectively? And what about SBI calls that would benefit from more than two (true) arguments?

> The existing content that specifies about the a7/a6 will continue to be there. I just specifying about the registers that don't need to be preserved across the SBI call here. As andrew mentioned earlier, Linux implementation expects all registers except a0 & a1 to be preserved. Hence the above suggestion.

A literal reading of your suggested wording requires a7 and a6 to be preserved, i.e. to have the same value when the ecall returns.  Do you really mean to specify this?

Also, do you really wish to limit the SBI interface to two parameters?  It seems that one of the changes introduced into early SBI proposals was to eliminate call by reference (pointer arguments); all arguments are now passed by value.  Limiting the ABI to two parameters will surely  cause that issue to raise its head again.

atishp04, 2022-02-02:
Why would preserving registers states (a2-a7) across SBI calls prevent them to be used ?

aswaterman, 2022-02-02:
There's no contradiction here.  a2-a7 are argument registers, but their values are preserved across the call.

https://github.com/torvalds/linux/blob/0809edbae347a224ca1b59fb8be1c2d54389c2c6/arch/riscv/kernel/sbi.c

konrad-schwarz, 2022-02-02:
> There's no contradiction here. a2-a7 are argument registers, but their values are preserved across the call.

> https://github.com/torvalds/linux/blob/0809edbae347a224ca1b59fb8be1c2d54389c2c6/arch/riscv/kernel/sbi.c

The code in the example does not derive any benefit from preserving the values of a2 -- a7: the life time of the GCC asm register variables ends with the enclosing function sbi_ecall().  Only a0 and a1 are referenced after the ecall instruction.  Callers of sbi_ecall() must assume that a2 -- a7 are clobbered as per the ABI.  For the purposes of backwards compatibility, a2 -- a7 might just as well be clobbered.

Note that the same is almost certainly true for the temporary registers t0 -- t6: any caller of sbi_ecall() must assume that these will be clobbered.   The code is so simple that no temporary registers will be needed across the ecall instruction -- what value could it hold?  Of course, ensuring this requires a clobbers clause of ` "t0", "t1", "t2", "t3", "t4", "t5", "t6" '.

Honestly, I though RISC-V was about technical elegance and minimalism, and "doing the right thing", not about (weak) backwards compatibility.

jrtc27, 2022-02-02:
Neither Linux nor FreeBSD have ever marked a2 - a7 as clobbered in their inline assembly. You therefore have absolutely no idea what the compiler's using them for, they could well be live out of the inline assembly, e.g. if sbi_ecall is inlined and one of the arguments is used later, it'll likely just reuse the argument register rather than waste instructions saving it elsewhere. Throwing around accusations about "(weak) backwards compatibility" is not helpful. We have many years of OS releases that absolutely rely on the current de-facto guarantees and changing that in the spec would be actively harmful, risking catastrophic breakage; that or nobody would actually follow the spec as a result and would instead reimplement the same behaviour as BBL and OpenSBI in order to maintain compatibility, resulting in the spec not being what anyone actually writes their code towards and being a waste of paper. The RISC-V ecosystem needs less of that not more, and arguing over a few registers in a firmware call interface this much is a total waste of time for everyone involved and not productive.

konrad-schwarz, 2022-02-02:
> You therefore have absolutely no idea what the compiler's using them for, they could well be live out of the inline assembly, e.g. if sbi_ecall is inlined and one of the arguments is used later, it'll likely just reuse the argument register rather than waste instructions saving it elsewhere.

As sbi_ecall() has external linkage, I would be very surprised if this were legal.

jrtc27, 2022-02-02:
> As sbi_ecall() has external linkage, I would be very surprised if this were legal.

sbi_ecall in the kernel is non-preemptible as it's not built with -fPIC and not marked weak, so copies of it can absolutely can be inlined within the same translation unit. But even if that doesn't in practice happen for Linux, it definitely does for FreeBSD, its sbi_call is a static inline function in a header.

atishp04, 2022-11-01:
Closing this issue as it is resolved now.

closed 2022-11-01 by atishp04