riscv-non-isa/riscv-sbi-doc #38: sbi_clear_ipi() no longer available in SBI v0.2

lbmeng, 2020-03-14:
I noticed that sbi_clear_ipi() is no longer available in SBI v0.2 spec.
I guess the reason why it was removed is because S-mode software could do such directly from S-mode without the SBI call.

I also see these RFENCE Extension functions in the v0.2 spec. But the jobs done by these RFENCE Extension functions could also be done by calling sbi_send_ipi() plus some additional work from S-mode software.

Could someone clarify this why we only removed sbi_clear_ipi() but still keep these RFENCE Extension functions?

atishp04, 2020-03-16:
For following reasons.

1. IPIs are still handled by SBI. You have to trap to M-Mode to execute IPI and return to S-mode to execute fence instruction which has higher latency than RFENCE SBI call. Here are some performance numbers I collected few months back.
https://patchwork.kernel.org/cover/10872349/

2. Without RFENCE SBI extension, hypervisors has to implement software emulated MMIO CLINT for Guest/VM. This is very slow because to inject IPI to N HARTs we need to do N MMIO writes where each MMIO write will be trap-n-emulated. 

Once we have S-mode IPI injection facility in CLINT, we can use just IPIs for RFENCE functions or If ISA supports fence instructions in S-mode, we can directly use that.

Even in that case, there may be some hardware which won't implement that and can rely on SBI extension. Linux kernel will be smart enough detect these at run time to choose the best approach to do fence calls.

atishp04, 2020-10-19:
Closing this issue as it is resolved.

closed 2020-10-19 by atishp04