riscv-non-isa/riscv-sbi-doc #46: Remote sfence vvma -- missing VMID argument

konrad-schwarz, 2020-04-27:
Section 5.3.2 of the Privileged Specification (Version 1.12) notes that the HFENCE.VVMA instruction is only operates on the guest identified by hgatp.VMID. (See also 5.5.3: "the effective ASID can be considered to be the combination of this VMID with the VS-level ASID.").

This is fairly obvious: a TLB holding guest translations needs to match against both the ASID, which is defined by the guest, and the VMID, which is defined by the hypervisor.

Hence,  hagtp.VMID is an implicit operand to the HFENCE.VVMA instruction.

However, the sbi_remote_hfence_vvma_asid and sbi_remote_hfence_vvma functions do not have an argument for the VMID.

I suggest adding the VMID as an additional argument to both functions, in the fifth respectively fourth position  Although this breaks the convention that semantically identical arguments are passed in identical registers, it makes the passthrough from a guest sbi_remote_fence_vma_asid and sbi_remote_fence simpler.  (Or you could add a dummy argument to sbi_remote_hfence_vvma function, which would allow you to keep the VMID argument in the fifth position in both functions.)

avpatel, 2020-04-27:
Yes, we should add VMID argument to SBI HFENCE.VVMA calls.

In fact, it's not too late since RISC-V H-Extension is not yet frozen and lot of Hypervisors are still in development.

Thanks for catching.

Can you send patch to tech-unixplatformspec@lists.riscv.org ?

Regards,
Anup

avpatel, 2020-04-28:
I recall now, we don't need the VMID argument because the SBI implementation can pickup-n-save the VMID from HGATP.

I guess we need to clarify the documentation.

Anyway thanks for pointing out.

Regards,
Anup

fintelia, 2020-04-28:
Does that mean that SBI will be baking in the assumption that VMID means the same thing between different harts?

aswaterman, 2020-04-28:
Which is a correct assumption for now... and hopefully in the long run, too, since ASID/VMID globality adds complexity but seems to have as many downsides as upsides.

avpatel, 2020-04-28:
Yes, VMID space is global for most open source hypervisors (KVM, Xen, Xvisor).

Regarding SBI HENCE.VVMA call, the semantics are same as local HFENCE.VVMA instruction which means the SBI implementation (i.e. M-mode firmware) should get current VMID from HGATP csr and use this VMID when doing HFENCE.VVMA on multiple HARTs.

The above mentioned SBI HENCE.VVMA call semantics should be documented in SBI spec.

Regards,
Anup

fintelia, 2020-04-28:
I know on x86, Linux uses a non-global ASID* scheme so it would be good to understand their design and evaluate whether it is something that RISC-V wants to rule out. The idea is to have core keeps a short history of the last 6 processes that ran on it, and for each memory map has a generation number and a bitmap of which cores it is currently installed on. Shootdowns are then only sent to cores which have the same page table active (if any), and otherwise the TLB gets flushed lazily on context switches if the generation number has changed.

Their argument is that translations are unlikely to linger in the TLB beyond a couple of context switches, so you can get almost all the benefit of ASIDs without the complexity and cost of global coordination to centrally manage ASID assignments. (The number of ASIDs is small compared to the possible number of running processes, and not even that much higher than the realistic limits on core count)

If a similar design were used by a hypervisor for VMIDs, then the desired semantics of remote_hfence_vvma would be to do what the documentation currently says: issue a hfence.vvma instruction without messing with hgatp at all.

_*Technically x86 calls them PCIDs_

aswaterman, 2020-04-28:
I'm firmly in the x86 camp on this one.  (My previous comment suggested otherwise, but that's because I misread a previous post.)  Avoiding global ASIDs/VMIDs at the ISA level saves a bunch of complexity, and sometimes even works out better for performance, since TLB shootdowns can be surgically targeted.

Whether SBI calls treat ASIDs as global is at least somewhat separable from the ISA decision, since OSes can simply choose not to use the SBI calls and use their own IPIs instead.

avpatel, 2020-04-28:
It's the Hypervisors that treat VMID space as global and not the SBI calls.

The semantics of SBI call are same as HFENCE/SFENCE instruction done on multiple HARTS. The SBI spec does not enforce any semantics on VMID or ASID.

To do remote TLB flushes locally within S-mode software will require dedicated S-mode CLINT OR or RISCV IPI extension for HARTs.

Regards,
Anup

aswaterman, 2020-04-28:
S-mode software can currently use the existing SBI calls for IPIs if it wants to do its own shootdowns.

avpatel, 2020-04-28:
Yes, SBI software can use SBI IPI calls for local remote TLB shoot down but it is slower than doing remote TLB flushes using SBI RFENCE calls. We have performance numbers to prove this as well.

The reason for better performance of SBI RFENCE is that CLINT IPI interrupt is wired to M-mode. When we do SBI IPI call, the M-mode software on target HART will first take interrupt in M-mode which then injects S-mode IPI. That's why we need a dedicated S-mode CLINT OR RISC-V IPI extension.

Regards,
Anup

avpatel, 2020-04-28:
The VMID handling in Xvisor RISC-V and KVM RISC-V are totally different.

The KVM RISC-V VMID handling is very similar to Linux x86 ASID handling. We don't have fixed VMID assigned to a Guest/VM but I won't call this approach 100% non-global because VMID allocation is done using a global cyclic counter (which is lockless). Whenever global cyclic VMID counter roll-over, we nuke TLBs of all HARTs and start re-assigning VMIDs again.

The Xvisor RISC-V has very simple approach of assigning fixed VMID to each Guest/VM. The reason for simple approach in XVISOR is that it an embedded hypervisor and it will typically run very few Guests/VM in resource constraint system.

Regards,
Anup

fintelia, 2020-04-28:
Your description of KVM RISC-V VMID handling still sounds global to me. Linux x86 by contrast can have a given ASID refer to a different process on every core. For instance, the first process that runs on each core is assigned ASID=1, the second is assigned ASID=2, etc. Once the first six have been assigned for a core, it uses an LRU scheme to reuse them. ASIDs 7-4095 are never used (except that if the KPTI Meltdown mitigations are enabled, each process actually gets two ASIDs)

avpatel, 2020-04-28:
Thanks for the details on x86 approach. I will study this more.

The KVM RISC-V approach of VMID allocation is designed to support large number of Guests/VMs so that we are not restricted by fixed number of VMID bits.

Irrespective to the approach taken for ASID/VMID allocation, we will always need remote TLB flushes because remote TLB flushes are also used when memory is mapped/unmapped for a process/guest/vm which happens at different rate based on workload.

In RISC-V systems, S-mode software doing remote TLB flushes locally makes sense only if S-mode software have way to do IPIs without M-mode software intervention.

Regards,
Anup

atishp04, 2022-11-01:
Closing this issue. Please reopen if it is not resolved.

closed 2022-11-01 by atishp04