riscv-non-isa/riscv-sbi-doc #47: sbi_clear_ipi deprecated?

konrad-schwarz, 2020-05-04:
The legacy extension `sbi_clear_ipi` has no equivalent in the revamped interface.  Has it been deprecated?

The architecture specification makes it clear that sip.SSIP is software writeable, hence no SBI function is needed:
> If implemented, SSIP is writable in sip. A supervisor-level software interrupt
> is triggered on the current hart by writing 1 to SSIP, while a pending supervisor-level software
> interrupt can be cleared by writing 0 to SSIP.

So `sbi_clear_ipi` *should* be deprecated (and the specification should make this explicit).

JohnAZoidberg, 2020-05-05:
Yes, that's the reason: https://github.com/riscv/riscv-sbi-doc/pull/27#issuecomment-589592759
Yes, this should be mentioned in the specification.

JohnAZoidberg, 2020-05-05:
See #38 also.

atishp04, 2020-05-29:
Closing this as it is fixed now.

closed 2020-05-29 by atishp04