riscv-non-isa/riscv-sbi-doc #54: How to add a new SBI call?

nylon7, 2020-05-28:
Hello all:

I implement some SBI call (non-platform-specific).

Both the Github issue & Patch talk about how to add SBI call too. (https://github.com/riscv/opensbi/issues/146) 
[PATCH] lib: Dynamic registration of SBI extensions http://lists.infradead.org/pipermail/opensbi/2020-January/000966.html

But the above discussions didn't touch on the standard process.

So I don't know how to define my SBI call's Extension ID. Or can I ignore it and define it myself?

JohnAZoidberg, 2020-05-28:
1. Think about whether it fits into an existing extension or else create a new extension
2. Think about all of the details of the behaviour and error cases
3. Document that in a patch for the SBI document
4. Send it to the (Mailing List](https://lists.riscv.org/g/tech-unixplatformspec) (You need to be subscribed to be able to post) and create a PR on GitHub
5. Argue for it ;)

It would be good if you implemented it in OpenSBI and sent patches there as well.

Please anyone correct me, if I forgot something.

I assume that you don't want to use the SBI call only for yourself but add it to the spec?

atishp04, 2020-05-29:
@JohnAZoidberg Thanks. You mentioned pretty much everything.

@nylon7 Regarding SBI extension ID, it depends on the usecase. If the extension is generic enough and should be part of the specification, you should pick something from the normal/experimental range and propose to the mailing list as suggested by Daniel.
If it is a platform specific SBI call, you may want to choose something from vendor extension range.

nylon7, 2020-06-01:
@JohnAZoidberg Thank you for your feedback.

@atishp04 So the conclusion is
1.non-platform-specific: I can follow what @JohnAZoidberg  is suggested 
2.platform-specific: If the platform has an extension ID, maybe I can direct- contribute my code to OpenSBI mailing list.

JohnAZoidberg, 2020-06-01:
The platforms don't have extension IDs. The extensions are for purposes, which are needed/useful on all platforms.
If you think your ecall is useful for people other than yourself, please discuss it on the mailing list.
If you let us know what it is, we can help you decide.

atishp04, 2020-06-01:
I agree with @JohnAZoidberg. If the new SBI extension is something that you can talk in public domain, I would suggest you to send a draft to the unix platform mailing list to start a discussion.

nylon7, 2020-06-02:
@atishp04 @JohnAZoidberg 

In fact, I have added platform-specific (AE350&extension ID:0x0900031E) and non-platform-specific

But I'm not sure contribution rule of both.

atishp04, 2020-10-19:
Closing this issues in absence of any new activity. Please create another issue if it is still unresolved.

closed 2020-10-19 by atishp04