riscv-non-isa/riscv-sbi-doc #55: How to decide Extension ID? For example: Timer Extension ID & IPI Extension ID

nylon7, 2020-06-04:
I was using git blame to search for "riscv-sbi.adoc" history message, and I find @atishp04 's commit(v0.1 SBI replacement extensions) which added  the Timer Extension ID and  IPI Extension ID.

I also searched in [Mailing List](https://lists.riscv.org/g/tech-unixplatformspec) discussion, but I can't find any Timer Extension ID and IPI Extension ID discussion, only system reboot extension discussion.

I want to know more about Extension ID's naming rules, and I was wondering if anyone can tell me where I can find more relevant information?

JohnAZoidberg, 2020-06-04:
It's the ASCII representation of the Extension:
- 0x54494D45 => 0x54, 0x49, 0x4D, 0x45 (TIME)
- 0x52464E43 => 0x52, 0x46, 0x4E, 0x43 (RFNC)
- ...

atishp04, 2020-10-19:
Closing this issues as it is resolved.

closed 2020-10-19 by atishp04