riscv-non-isa/riscv-sbi-doc #56: Extension for controlling performance counters

msullivan, 2020-09-02:
The privileged ISR document says "The implementation should provide a mechanism to modify the counter values." (from supervisor mode). Would it make sense to add a standard SBI extension for setting counter values and their configuration registers?

aswaterman, 2020-09-02:
This is being discussed on one of the RISC-V technical committee mailing lists https://lists.riscv.org/g/tech-privileged/topic/75675990 -- membership required, but free

avpatel, 2020-09-02:
The PMU enhancements for RISC-V privilege spec to support overflow interrupt and event filtering is being discussed at RISC-V technical committee mailing list.
(Refer, https://lists.riscv.org/g/tech-privileged/topic/75675990)

The SBI PMU Extension proposal is being discussed in RISC-V UnixPlatformSpec mailing list. The proposed SBI PMU extension considers existing HW with current PMU and also considers future HW with enhanced PMU (mentioned above).
(Refer, https://lists.riscv.org/g/tech-unixplatformspec/message/197)

As-per latest SBI PMU extension proposal, the SBI_PMU_COUNTER_START() call can set the counter at time of starting event counting.

Regards,
Anup

atishp04, 2020-10-19:
I think this issue is resolved. Please create another if there are any issue with PMU extension.

closed 2020-10-19 by atishp04