riscv-non-isa/riscv-sbi-doc #78: How should I get my SBI implementation ID?

retrhelo, 2021-08-08:
I implemented my own SBI implementation for my team's project, and wonder how I can get my SBI implementation ID. Is there a test, registry or something else that I should complete first before I can get my own specific ID for my SBI?
[Here](https://github.com/retrhelo/psicasbi) is my SBI implementation, written in Rust. It implements Base, Time, sIP extensions, and a subset of Legacy ones. I'm now using it on my team's project to replace the old version of RustSBI that's now in use. The PsicaSBI can now run on k210 platform and QEMU.

nylon7, 2021-08-09:
@retrhelo 

maybe you can reference this [issue](https://github.com/riscv/riscv-sbi-doc/issues/55)

retrhelo, 2021-08-09:
I see. But this rule seems to be for Extension ID. Is SBI ID following the same rule?

nylon7, 2021-08-09:
@retrhelo 

in my understand

Document that in a patch for the SBI document Send it to the [Mailing List](https://lists.riscv.org/g/tech-unixplatformspec) (You need to be subscribed to be able to post) and create a PR on GitHub
e.g [PATCH] [Add system reboot extension](https://lists.riscv.org/g/tech-unixplatformspec/message/7?p=,,,20,0,0,0::recentpostdate%252Fsticky,,system+reboot,20,2,20,72670816)

retrhelo, 2021-08-09:
All Right, I'll check it out. Thanks. :)

atishp04, 2022-02-04:
No activity on this issue. Closing this one. Please reopen if you have still some issues.

closed 2022-02-04 by atishp04