riscv-non-isa/riscv-sbi-doc #81: how do arguments of sbi function passed to SEE?

agrobman, 2021-11-04:
Sorry, maybe I missed something, but how do the sbi functions pass their parameters to SEE.

the a7 is used to pass EID
a6 is used to pass FID,

what register is used to pass stime_value in sbi_set_timer function, for ex?
Are these a0,a1 ... as ABI mandate?

repnop, 2021-11-18:
Since no one has answered you yet, its spelled out in [this section](https://github.com/riscv-non-isa/riscv-sbi-doc/blob/5b856c26dc6b12d5b15ee4f93651592803ac4c50/riscv-sbi.adoc#binary-encoding) of the specification:
> This binary encoding matches the RISC-V Linux syscall ABI, which itself is based on the calling convention defined in the RISC-V ELF psABI.

Now if we look at the [calling convention specification](https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/32675540208d6f144dff0427ea49a418aceb617e/riscv-cc.adoc):

| Name      | ABI Mnemonic | Meaning                | Preserved across calls?
|---|---|---|---
| x0        | zero         | Zero                   | -- (Immutable)
| x1        | ra           | Return address         | No
| x2        | sp           | Stack pointer          | Yes
| x3        | gp           | Global pointer         | -- (Unallocatable)
| x4        | tp           | Thread pointer         | -- (Unallocatable)
| x5 - x7   | t0 - t2      | Temporary registers    | No
| x8 - x9   | s0 - s1      | Callee-saved registers | Yes
| x10 - x17 | a0 - a7      | Argument registers     | No
| x18 - x27 | s2 - s11     | Callee-saved registers | Yes
| x28 - x31 | t3 - t6      | Temporary registers    | No

So arguments are passed to the SBI implementation in order of `a0` to `a5`, with `a6` and `a7` being overridden by this specification to specifically be the FID and EID, respectively, for calls.

agrobman, 2021-11-18:
thanks for clarification..

closed 2021-11-18 by agrobman