riscv-non-isa/riscv-sbi-doc #82: Inaccuracies of legacy SBI specification

jrtc27, 2021-11-14:
BBL is the canonical implementation of the legacy spec, since the spec is meant to be a formalisation of BBL's behaviour; OpenSBI somewhat agrees with it, but not entirely, and the spec disagrees with both even in places where the two implementations agree.

## Extension: Set Timer (EID #0x00)

This returns a long of value 0 in both BBL and OpenSBI.

## Extension: Console Putchar (EID #0x01)

This returns a long of value 0 in both BBL and OpenSBI.

## Extension: Console Getchar (EID #0x02)

This returns 0 in BBL when no supported UART is detected, whereas for OpenSBI it returns -1, just like if the UART's receive buffer is empty. OpenSBI's behaviour makes more sense, but is technically wrong as BBL is canonical. Both also return a long, not an int.

## Extension: Clear IPI (EID #0x03)

This returns `old_mip & MIP_MSIP` in BBL, and 0 in OpenSBI.

## Extension: Send IPI (EID #0x04)

This returns a long of value 0 in BBL, but can return errors in OpenSBI.

## Extension: Remote FENCE.I (EID #0x05)

This returns a long of value 0 in BBL, but can return errors in OpenSBI.

## Extension: Remote SFENCE.VMA (EID #0x06)

This returns a long of value 0 in BBL, but can return errors in OpenSBI.

## Extension: Remote SFENCE.VMA with ASID (EID #0x07)

This returns a long of value 0 in BBL, but can return errors in OpenSBI.

## Extension: System Shutdown (EID #0x08)

Both agree this never returns, but were it to it would return 0 in OpenSBI (BBL is unspecified, there is no return statement for mcall_shutdown).

---

When it comes to SBI calls that access memory on behalf of the supervisor, it's also totally undocumented that both will redirect page and access faults to the supervisor so it looks like the ecall faulted.

The important point though is that every single one of these returns a long, and thus the "This is the only SBI call in the legacy extension that has a non-void return type." note for getchar is completely false. Any unknown call is handled by BBL as returning -ENOSYS, which is nonsense due to being an OS-specific errno that is wrong to use in firmware and should have been an SBI constant, but does "work" in the sense that it's a negative value and thus an error. This also means "An ECALL with an unsupported SBI extension ID (EID) or an unsupported SBI function ID (FID) must return the error code SBI_ERR_NOT_SUPPORTED." does not hold for BBL.

atishp04, 2022-02-16:
The objective of the legacy version of the specification was to keep the old SBI v0.1 specification as it is. BBL/OpenSBI implementation might have differed over time. But I don't think we need any change in the specification to match the implementation. If there is any change required, it should be other way around.

I don't think anybody is using SBI v0.1 right now and it's disabled by default from Linux kernel as well.

jrtc27, 2022-02-17:
That makes no sense. If the specification for SBI v0.1 does not match what any legacy implementation does then it is totally useless and you might as well not bother specifying it; at least then it wouldn't be actively misleading.

jrtc27, 2022-02-17:
Linux distros will likely enable v0.1 support in order to work with BBL. FreeBSD does and has no option to disable it.

atishp04, 2022-02-17:
> Linux distros will likely enable v0.1 support in order to work with BBL. FreeBSD does and has no option to disable it.

Why does distros care about running v0.1 with BBL when it is lacking many features ?

jrtc27, 2022-02-17:
Because people have systems that are set up to use BBL and it works? Sure OpenSBI has fancy features but if all you need is to boot a dev board then why would the user care. We're still using it because we haven't got round to porting OpenSBI, we have far more important things than that to do.

aswaterman, 2022-02-17:
I agree with @jrtc27 on both counts.  Pretending SBI v0.1 isn't important is something of a fiction, because it does work and people do use it.  (Especially since Linux is not the universe.)  And given that, it makes sense for the SBI v0.1 spec to accurately reflect the SBI v0.1 implementations.

avpatel, 2022-02-17:
@jrtc27 can you please send a PR for the changes in SBI v0.1 text ? The SBI v1.0 is in final stages of ratification cycle so this will save time.

Regards,
Anup

jrtc27, 2022-02-17:
I don't have time to do this, please do it yourselves. I already gave you a detailed list of the problems with the current specification, so it should be straightforward. It's not my fault you chose to ignore this for three months whilst pushing forward with ratification efforts.

aswaterman, 2022-02-17:
@jrtc27 Hmm, BBL returning 0 for console_getchar if there's no UART is a bug introduced by Palmer in 2017, since BBL returns -1 if there _is_ a UART but there's no input data.  And of course the Linux driver also assumes a return value of 0 to mean that the character 0 was read from the input source.  This can be retconned as a BBL "feature" that, if it can't find a UART, it falls back on a UART that supplies an infinite source of 0 characters.  I know it's eye-roll-inducing, but, you know...

jrtc27, 2022-02-17:
We could also just change it :) given -1 is already returned in the UART-without-data case it shouldn't break anything. FreeBSD filters out 0 bytes (which sucks if you actually want to send 0, but works around this issue) whereas Linux won't, so Linux probably gets very confused if there is no SBI UART as it stands...

jrtc27, 2022-02-17:
But I don't think it's a huge deal; if BBL has no UART then something's gone wrong and you should fix that anyway

aswaterman, 2022-02-17:
@jrtc27 I was actually hoping you'd suggest changing it :-)

repnop, 2022-02-21:
sorry if this is the wrong issue to ask in -- but is the ABI for the legacy extensions the same as v0.2 and onward, minus the unused registers? specifically, I'm working on a Rust library to interface with SBI and its not clear to me whether this still holds true for the legacy spec:
> All registers except `a0` & `a1` must be preserved across an SBI call by the callee.

since the functions don't follow the same return structure as v0.2+

in either case, its probably worth explicitly documenting what the ABI for the legacy extensions so its more obvious for future readers, I would think

aswaterman, 2022-02-22:
@repnop The more precise statement is that all registers except `a0` are preserved. (Right, @jrtc27?  It appears both BBL and OpenSBI do this.)

atishp04, 2022-02-22:
@aswaterman @repnop Anup has updated the PR #91 to add this.

repnop, 2022-02-22:
awesome, thanks for the clarification :+1:

jrtc27, 2022-02-22:
> @repnop The more precise statement is that all registers except `a0` are preserved. (Right, @jrtc27? It appears both BBL and OpenSBI do this.)

That's my understanding too, yes

atishp04, 2022-11-01:
closing this issue as it is resolved now.

closed 2022-11-01 by atishp04