riscv-non-isa/riscv-sbi-doc #92: `sbi_remote_sfence_vma` and co.: what does it mean for `start_address` or `size` to be invalid?

repnop, 2022-02-28:
Currently, the documentation around the various `rfence` functions states that they're able to return `SBI_ERR_INVALID_ADDRESS` whenever `start_address` or `size` is "invalid", however its not clear how those values would be considered invalid. OpenSBI doesn't seem to check for any kind of invalid values:
https://github.com/riscv-software-src/opensbi/blob/013ba4ef3d94de67d040376535131012134ed54f/lib/sbi/sbi_tlb.c#L79-L98
```c
void sbi_tlb_local_sfence_vma(struct sbi_tlb_info *tinfo)
{
	unsigned long start = tinfo->start;
	unsigned long size  = tinfo->size;
	unsigned long i;

	sbi_pmu_ctr_incr_fw(SBI_PMU_FW_SFENCE_VMA_RCVD);

	if ((start == 0 && size == 0) || (size == SBI_TLB_FLUSH_ALL)) {
		tlb_flush_all();
		return;
	}

	for (i = 0; i < size; i += PAGE_SIZE) {
		__asm__ __volatile__("sfence.vma %0"
				     :
				     : "r"(start + i)
				     : "memory");
	}
}
```

`size` seems to be the size of the region in bytes to `SFENCE.VMA` (which I think could be made more clear as well, as in my opinion its easy to confuse `size` to mean either a byte size or a page size, even though the latter doesn't make sense) and OpenSBI does nothing except check for some sentinel values for some specific behaviors (can those sentinel values be relied on or is it implementation specific behavior?). Meanwhile, `SFENCE.VMA` can't trigger an exception as per the spec:
> If the value held in _rs1_ is not a valid virtual address, then the `SFENCE.VMA` instruction has no effect. No exception is raised in this case.

And I doubt that any implementation is going to attempt to walk page tables to determine if the address is valid w.r.t virtual memory mappings.

The only error case I can think of is passing a `size` that's not a multiple of 4 KiB, but I'm not sure if there's sufficient reason to actually do so other than being somewhat paranoid about parameter correctness.

aswaterman, 2022-02-28:
IMO, there should be no errors, by definition; the range flushed should be [round_down(addr, 4096), round_up(addr + size, 4096) - 1].  This follows the definition of the `sfence.vma` instruction itself, which explicitly permits invalid address arguments.

avpatel, 2022-03-09:
The SBI_ERR_INVALID_ADDRESS was mainly for alignment and sanity checks.

Existing SBI implementations (e.g. OpenSBI, KVM, Xvisor, etc) generally don't return this error.

Regards,
Anup

atishp04, 2022-11-01:
Closing this issue. Please reopen if it is not resolved.

closed 2022-11-01 by atishp04