riscv-non-isa/riscv-sbi-doc #97: `sbi_hart_start` doesn't return `SBI_ERR_ALREADY_STARTED` when a hart is already started

repnop, 2022-03-13:
When a given hart is already started and `sbi_hart_start` is called on it again, the spec says that it'll return `SBI_ERR_ALREADY_AVAILABLE`, but since the addition of the PMU extension, there's a `SBI_ERR_ALREADY_STARTED` error that exactly conveys the meaning of the error and would be more suitable, in my opinion.

atishp04, 2022-11-01:
We just wanted to be explicit in this case as "available" means something else in the PMU context. For HSM extension, it indicates that the given hart is already available in S-mode (e.g started). Otherwise. the hart is idling in M-mdoe.

However, the HPM counters are always available in S-mode. PMU extension allows it start/stop from S-mode.

atishp04, 2023-03-27:
closing this issue due to inactivity. Please reopen if above answer is not sufficient.

closed 2023-03-27 by atishp04