riscv-non-isa/riscv-sbi-doc #111: Progress of RISC-V SBI calling convention maps to the ARM PSCI calling convention

gagachang, 2023-05-08:
I found the following slides shows the compaison between SBI and ARM PSCI:
https://lists.riscv.org/g/tech-unixplatformspec/topic/risc_v_sbi_vs_arm_psci/77906648

And there is a topic:
```
How the RISC-V SBI calling convention maps to the ARM PSCI calling convention (and vice-versa)
```

Further more, ARM SMCCC has four registers `W0 ~ W3` used to be return values, althought SBI only has `a0 and a1`.

Is there any progress or materials to standarize this topic?

atishp04, 2023-08-28:
There is no plan to make any changes to return arguments and extend it more registers similar to PSCI. The slides reference above was presented to indicate that we can't reuse most of the PSCI spec for the reasons describe there.

atishp04, 2023-10-10:
Can we close this ? @gagachang

gagachang, 2023-10-10:
Sure, I am closing it.

closed 2023-10-10 by gagachang