riscv-non-isa/riscv-sbi-doc #136: Base extension function names do not match end-of-section function listing

blueseaslug, 2023-12-16:
Hello, forgive me if I am incorrect, but I think there is an inconsistency with the functions listed in the Base extension (EID 0x10).

The functions `sbi_get_spec_version(void)` (FID 0x0), `sbi_get_impl_id(void)` (FID 0x1), and `sbi_get_impl_version(void)` (FID 0x2) are documented as usual in the Base extension description.

However, in Table 1 at the end of the section, these functions are listed as `sbi_get_sbi_spec_version`, `sbi_get_sbi_impl_id`, and `sbi_get_sbi_impl_version` (note the extra "`sbi`" in the middle of each of them).

These are the  only functions that differ between their in-section definition and the whole-section listing at the end of a given section. I cannot see how to satisfy both definitions in a consistent way. For these two reasons I think they are meant to be the same function, is this correct? If so, which is the right definition?

Thanks

atishp04, 2023-12-18:
@blueseaslug : Thanks for pointing out the discrepancies in the name. It was just a typo that went unnoticed for long. No need for additional "sbi" in the function name.

blueseaslug, 2023-12-21:
Got it thank you

closed 2023-12-21 by blueseaslug