riscv-non-isa/riscv-sbi-doc #141: Regarding Unofficial hartid Transmission from Machine Mode

mkei-154, 2024-02-06:
Hi,

I am involved in the development of an operating system that operates in Supervisor Mode. As you may be aware, it is not possible to directly retrieve hartid in Supervisor Mode. To obtain hartid, one must access the mhartid CSR, and thus, hartid can only be acquired in Machine Mode.

Upon reviewing the Linux kernel and OpenSBI, it is apparent that hartid is typically set in the a0 register, becoming an argument received by the booted entity during startup.

However, this is not a defined "specification." Depending on the implementation, it is possible that hartid may not be passed as a startup argument. Nevertheless, many implementations would likely fail to function properly without passing hartid as a startup argument. In essence, this becomes a "de facto specification." Why is it not officially specified that hartid must be passed during transitions between these different modes? I consider this to be a highly precarious situation.

If this is the wrong forum to raise this issue, I am sorry, but please let me know where to raise it.

SiFiveHolland, 2024-02-06:
This is explicitly documented for non-boot harts as part of the HSM extension ([Table 2. HSM Hart Start Register State](https://github.com/riscv-non-isa/riscv-sbi-doc/blob/master/src/ext-hsm.adoc#table_hsm_hart_start_regs)). However, I agree the specification is missing a description of the register state when first entering supervisor mode on the boot hart.

mkei-154, 2024-02-09:
I believe OpenSBI should explicitly state the passing of hartid during the transition to the Next Boot Stage of Boot hart. Personally, I recommend passing hartid during every Boot Stage transition as part of RISC-V, but I understand that it might be challenging.

aswaterman, 2024-02-09:
I agree with OP that this must be part of the spec. It’s not just a convention. It was specified as part of the earlier SBI specs and somehow disappeared along the way.

Given it was once canonical, and is relied upon by OSes in practice, it should be re-added. I’d approve a PR that does so.