riscv-non-isa/riscv-sbi-doc #169: Unclear how to implement uint64_t on RISC-V 32 (see for example chapter 5.1)

sfiedler0, 2024-10-07:
For the type `long`, we know that it is architecture-dependent (RISC-V 32 gets 4 bytes, RISC-V 64 gets 8 bytes).  
But when we have a uint64_t, there is zero information in the chapter about binary encoding on when to just use two arguments (one with the low and one with the high part), or what to do else. 

Also, a bit vice-versa: some functions, like console write (chapter 12.1), take `base_addr_lo` and `base_addr_hi` parameters. On RISC-V 64, wouldn't it be possible to just write one `unsigned long` for the 64-bit address instead of taking two parameters?

SiFiveHolland, 2024-10-07:
For uint64_t parameters on riscv32, I believe the SBI specification would defer to [the RISC-V calling convention defined in the PSABI](https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/master/riscv-cc.adoc#integer-calling-convention):

> Scalars that are 2×XLEN bits wide are passed in a pair of argument registers, with the low-order XLEN bits in the lower-numbered register and the high-order XLEN bits in the higher-numbered register.

Address parameters cannot be `unsigned long` because riscv32 supports 34-bit physical addresses. They could have been `uint64_t`, but it's just a matter of existing functions being too late to change now, and new functions doing the same for consistency.

SiFiveHolland, 2024-10-07:
The calling convention is clarified by https://lists.riscv.org/g/tech-prs/topic/patch_clarify_some_calling/108873941

Note that the spec does provide some rationale for the address encoding:

> It is recommended that a memory physical address passed to an SBI function
should use at least two `unsigned long` parameters to support platforms
which have memory physical addresses wider than XLEN bits.

SiFiveHolland, 2024-10-17:
Resolved by 1ecc7ece9052de9d22acb415055c8d1a9f50a6f4

closed 2024-10-17 by atishp04