riscv-non-isa/riscv-sbi-doc #188: Feedback on SBI 3.0

ved-rivos, 2024-12-10:
DBTR:
1. "The configuration of each debug trigger is expressed by three entities" ->
   "The configuration of each debug trigger is expressed by three parameters" ->

2. sbi_debug_set_shmem - is it allowed for the shared memory to be
   discontigous i.e. wrap around at the boundary of the maximum supported
   physical address?

3. sbi_debug_read_triggers - does the trig_state get sanitized in any form?
   Why is desired to return the hardware index? For SBI provided by a 
   hypervisor, is the expectation that the VU/VS refer to the guest of the
   guest hypervisor? Is it an error to specify non-zero VU/VS privileges
   H extension is not supported.
  
4. The usage of the mask in functions such as sbi_debug_uninstall_triggers
   is not specified.

MPXY:

1. MSI Interrupt -> MSI

2. sbi_mpxy_write_attributes - description is copied from sbi_mpxy_read_attributes
   and needs update.

3. sbi_mpxy_send_message_with_response - "After sending the message, this
   function waits for SBI implementation for the message response." This text
   does not read right.

4. "Some messages may require sending multiple times for complete data transfer "
   Is the intent to say "send data using multiple messages" or "Send multiple
   messages for complete data transfer"? This sentence should be reworded.

5. "If any notification indication mechanism" -> "If no notification mechanism"?
    Reword the second non-normative note on page 97

6. "In the shared memory, 16 bytes starting from offset 0x0 are used for this
   state data." Is the intent that the 16 bytes are used to return the state
   data and not used for holding this state data? 

FWFT:

1. FWFT states "control behavior of firmware features". However, the types
   related mostly to controlling hardware features. The description should be
   updated to better reflect the extensions purpose.

2. When supervisor mode exists, medeleg must exist. The description "if medeleg
   is present" is redundant as supervisor mode is required for SBI.

3. Pointer masking tag length -> pointer masking length

4. "for supervisor mode" is redundant in description of DOUBLE_TRAP.

5. If an implementation supports PMLEN of 7 and 16 and FWFT may pick a 
   PMLEN >= N then it seems impossible to set PMLEN of 7 if the implementation
   always picks 16. Is that intentional? Why?

6. sbi_fwft_get does not provide a way to read back the flags such as the LOCK.

SSE:

1. "global software event is a system event and can be handled by any
     hart." this may want to be more constrained to say "participating harts"

2. Should there be two RAS events - to denote high and low priority?

3. EventID based priority - this would always favor the lower numbered 16-bit
   group. Is that intentional?

4. Is a partial success allowed when write of SSE an event attribute fails?

5. Is handler entry PC checked for validity? Are there rules for supervisor
   when its switching to/from Bare virtual memory system mode.

6. When a SSE is registered, is it in masked state? Should mask/unmask check
   for being in REGISTERED state or ENABLED state?

7. This section is missing guidance and rules on implementing SSE handlers. 
    For instance, is it allowed for an SSE handler to cause an exception. 

8. Have the interactions of SSE with double trap been considered? Is it legal
    to cause an SSE when SDT is 1 for reasons other than notifying occurrence
    of a double trap?

atishp04, 2024-12-17:
> DBTR:
> 
> "The configuration of each debug trigger is expressed by three entities" ->
> "The configuration of each debug trigger is expressed by three parameters" ->
> 

Done.

> sbi_debug_set_shmem - is it allowed for the shared memory to be
> discontigous i.e. wrap around at the boundary of the maximum supported
> physical address?
> 

Any shared memory in the SBI spec is guided by chapter 3.2. Shared memory physical address range parameter. Discontiguous memory is not allowed. If shared memory crosses maximum supported physical address then it is technically not accessible to supervisor software so we should return SBI_ERR_INVALID_ADDRESS
Please let us know if you think more clarification is necessary in 3.2

> sbi_debug_read_triggers - does the trig_state get sanitized in any form?

Some of the RISC-V implementations require actual hardware trigger index to start/stop trace. Upon request from such RISC-V implementations, we have kept an optional hardware index trigger state which SBI implementation may or may not provide.

> Why is desired to return the hardware index? 

Some of the RISC-V implementations require actual hardware trigger index to start/stop trace. Upon request from such RISC-V implementations, we have kept an optional hardware index trigger state which SBI implementation may or may not provide.

> For SBI provided by a
> hypervisor, is the expectation that the VU/VS refer to the guest of the
> guest hypervisor? Is it an error to specify non-zero VU/VS privileges
> H extension is not supported.
> 

Yes, that is correct. Specifying VU/VS privileges in trig_tdata1 when H-extension is not support is clearly an invalid configuration and sbi_debug_install_triggers() returns SBI_ERR_INVALID_PARAM.

> The usage of the mask in functions such as sbi_debug_uninstall_triggers
> is not specified.
> 

The SBI implementation finds a set of trigger IDs with a combination of base and mask. This allows supervisor mode software to uninstall/enable/disable multiple triggers at once. It can invoke these functions multiple times with different base to set trigger IDs > `unsigned long` as well.

Clarifications to each function along these lines added. See PR #

> MPXY:
> 
> MSI Interrupt -> MSI
> 

Done.

> sbi_mpxy_write_attributes - description is copied from sbi_mpxy_read_attributes
> and needs update.
> 

Overall layout of the text is the same as sbi_mpxy_read_attributes() but there are differences. Can you let us know which part of text needs to be updated ?

> sbi_mpxy_send_message_with_response - "After sending the message, this
> function waits for SBI implementation for the message response." This text
> does not read right.
> 

Fixed. See PR [PR](https://github.com/riscv-non-isa/riscv-sbi-doc/pull/190)

> "Some messages may require sending multiple times for complete data transfer "
> Is the intent to say "send data using multiple messages" or "Send multiple
> messages for complete data transfer"? This sentence should be reworded.
> 

Fixed. See [PR](https://github.com/riscv-non-isa/riscv-sbi-doc/pull/190)

> "If any notification indication mechanism" -> "If no notification mechanism"?
> Reword the second non-normative note on page 97
> 
[... 71 lines cut ...]
Nope. It's supervisor software's responsibility to pass a valid address. As SBI implementation shouldn't care or access that address at anytime. Thus, if it is an invalid address, it will fault in Supervisor mode after jumping to it.

> Are there rules for supervisor
> when its switching to/from Bare virtual memory system mode.
> 

It is the supervisor software’s responsibility to either enable/disable event OR re-register a different handler when switching to/from bare virtual memory mode.

> When a SSE is registered, is it in masked state? Should mask/unmask check
> for being in REGISTERED state or ENABLED state?
> 

Each SSE have an “enabled” state and no mask/unmask state (please refer to the “Software Events State”. The SSE state transitions are independent of the HART SSE mask/unmask state. This means we don’t need to check HART mask/unmask state for registering events. Currently, upon CPU power-down we mask all SSE events for the CPU and unmask all SSE events upon CPU power-up. Same applies to when CPU goes in non-retentive suspend and comes back.

> This section is missing guidance and rules on implementing SSE handlers.
> For instance, is it allowed for an SSE handler to cause an exception.
> 

Taking trap when handling SSE event in supervisor mode can be catastrophic for supervisor software. We will add some non-normative text.

> Have the interactions of SSE with double trap been considered? Is it legal
> to cause an SSE when SDT is 1 for reasons other than notifying occurrence
> of a double trap?

Taking an SSE has higher priority than taking a trap/interrupt from a supervisor perspective. Each SSE has its own handler and context from a supervisor's perspective, so supervisor software can take SSE even while it was in a low-level trap handler saving/restoring registers.

atishp04, 2024-12-17:
@avpatel @hschauhan @SiFiveHolland @clementleger @pathakraul @jones-drew : Please add anything if I missed something.

pathakraul, 2024-12-17:
I checked the MPXY related changes and they looks good to me

ved-rivos, 2024-12-17:
> Overall layout of the text is the same as sbi_mpxy_read_attributes() but there are differences. Can you let us know which part of text needs to be updated ?

"Supervisor software MUST read the message protocol specific attributes via separate call to this function ...." This function is used to write the attributes and not read the attributes.
"Upon calling this function the message channel attribute values are returned starting from the offset 0x0 in the shared memory of the calling hart where the value to be written in attribute with attribute_id =base_attribute_id + i is at the shared memory offset 4 * i." This function should be updating the attribute values and not returning attribute values.

Do these sentences need to be updated?

ved-rivos, 2024-12-17:
> "In the shared memory, 16 bytes starting from offset 0x0 are used for this
state data." Is the intent that the 16 bytes are used to return the state
data and not used for holding this state data?

> Yes.

Then "used for this state data" would want to be updated to "used to return this state data" or similar.

ved-rivos, 2024-12-17:
> We are not sure if we follow your question here. The firmware feature allows to set the PMLEN value >=N. When you say "implementation always picks 16" - Do you mean SBI implementation or hardware ?

The specification states: "Enable pointer masking for supervisor-mode with PMLEN >= N." This implies that the SBI implementation when invoked with a value of N may choose to enable pointer masking with PMLEN = N or PMLEN > N. Is that the intent to provide the flexibility for the SBI implementation to pick either the specified value of N or any value greater than N?

ved-rivos, 2024-12-17:
> Yes. There is no mechanism to return the state of the flags. Do you envision a use case where it will be useful ? If it is just for informational purposes for supervisor mode software, we can introduce another explicit return code. Currently, SBI_ERR_DENIED is returned if the feature is locked.

Please consider whether usages like kexec may be surprised by error denied.

atishp04, 2024-12-17:
> Then "used for this state data" would want to be updated to "used to return this state data" or similar.

Fixed in the PR #190

atishp04, 2024-12-17:
> The specification states: "Enable pointer masking for supervisor-mode with PMLEN >= N." This implies that the SBI implementation when invoked with a value of N may choose to enable pointer masking [...]

Thanks for the clarification. Yes. If the value is set to <=7, it will first try to set 7. If failed for some reason, it will try to set 16. 
@SiFiveHolland : Please add anything else if required.

atishp04, 2024-12-18:
> Please consider whether usages like kexec may be surprised by error denied.

kexec path will invoke hart stop via SBI HSM call. The SBI implementation will unlock any feature that was previously locked. Am I missing any other use case ?

SiFiveHolland, 2024-12-18:
> Thanks for the clarification. Yes. If the value is set to <=7, it will first try to set 7. If failed for some reason, it will try to set 16. @SiFiveHolland : Please add anything else if required.

Yes, it is exactly the intention to provide flexibility to the SBI implementation. The assumption is that software written for a particular PMLEN value will function transparently with a larger PMLEN. It is valid for hardware to support PMLEN = 16 but not PMLEN = 7, and the current API abstracts this away from S-mode software. This also provides transparent forward compatibility for whatever meaning is assigned to the reserved PMLEN encoding. The OpenSBI algorithm of trying PMLEN = 7 first is not guaranteed by the spec.

SiFiveHolland, 2024-12-18:
> kexec path will invoke hart stop via SBI HSM call. The SBI implementation will unlock any feature that was previously locked. Am I missing any other use case ?

The "boot hart" of the new kernel will not go through HSM hart stop. I still think the current behavior is okay; if S-mode chooses to lock a feature, then it must be prepared for the consequences (e.g. maintaining a valid shadow stack through the kexec path).

ved-rivos, 2024-12-18:
The text as written states: "Enable pointer masking for supervisor-mode with PMLEN >= N". Consider an implementation that always picks "16" irrespective of whether N was 7 or 16. In such an implementation, there no way to select 7 even when the hardware supports it. Is picking 16 when OS requested 7 transparent for all OS's. Would some OSs rather not run with PM when their requested length is not supported by hardware? Would a Sv57 OS that invoked this call when it had data mapped into high addresses be surprised when a PMLEN wider than what it requested becomes active when this call returns?

ved-rivos, 2024-12-18:
> kexec path will invoke hart stop via SBI HSM call. The SBI implementation will unlock any feature that was previously locked. Am I missing any other use case ?

Should this text clarify that sbi_hart_stop should be considered hart reset?
> If provided, once set, the feature value can no longer be modified until:
> - hart reset for feature with local scope
> - system reset for feature with global scope

SiFiveHolland, 2024-12-18:
> The text as written states: "Enable pointer masking for supervisor-mode with PMLEN >= N". Consider an implementation that always picks "16" irrespective of whether N was 7 or 16. In such an [...]

Right. A larger than requested PMLEN is not transparent when the mask bits overlap with valid virtual address bits. Since the SBI implementation cannot know which addressing mode S-mode will use, the SBI implementation would only be allowed to choose a larger PMLEN if doing so doesn't overlap with the hardware's widest addressing mode. Either that, or we should require that PMLEN = N all the time.

atishp04, 2024-12-18:
> mode

> The "boot hart" of the new kernel will not go through HSM hart stop. I still think the current behavior is okay; if S-mode chooses to lock a feature, then it must be prepared for the consequences [...]

I think it is already taken care of in shadow stack patches. Cc: @deepak0414 for confirmation.

ved-rivos, 2024-12-18:
> I think it is already taken care of in shadow stack patches. Cc: @deepak0414 for confirmation.

The primary question was about the ability to read back that a feature is locked. Not being able to unlock it seems fine. However, presently software has no way to determine a) whether it is locked and b) reason why a feature that is being enumerated is being denied. If software could read the status of that the feature is locked it may not get surprised. Is this an issue?

atishp04, 2024-12-18:
> The primary question was about the ability to read back that a feature is locked. Not being able to unlock it seems fine. However, presently software has no way to determine a) whether it is locked [...]

via another API(e.g get_flag_status) or just as a explicit error code as SBI_ERR_LOCKED ?

If there are multiple flags define, the explicit API would fit well. It may be an an overkill for just for 1 flag as we are not sure if we will have many more flags in the future.

ved-rivos, 2024-12-18:
> via another API(e.g get_flag_status) or just as a explicit error code as SBI_ERR_LOCKED ?

Will defer that to the TG to decide the right way.

atishp04, 2024-12-18:
> > via another API(e.g get_flag_status) or just as a explicit error code as SBI_ERR_LOCKED ?
> 
> Will defer that to the TG to decide the right way.

Sure. We will discuss in the TG in the next meeting.

deepak0414, 2024-12-30:
First thing `kexec` flow will do is that it will take hart `satp` to `bare` mode.
And that will effectively break shadow stack. If menvcfg.SSE = 1, then all shadow stack instructions will
start raising store/AMO  access-fault exceptions. 
So `kexec` flow has to ensure that menvcfg.SSE = 0 (thus not have shadow stack FWFT locked),
so that shadow stack enabling is re-established.

On topic of locking shadow stack with respect to `kexec`.
Perhaps locking shadow stack with `kexec` is not ideal for following reason

"""
If an attacker has found a vulnerability in kernel to perform a ROP, they can perform a `kexec` on a kernel
which wasn't compiled with shadow stack instructions and thus completely bypassing shadow stack protections.
Thus going out of the way to ensure that locking of shadow stack works with `kexec` may not be useful anyways.
"""

So it should be a kernel policy to ensure that if kernel is enabled to perform `kexec` then shadow stack FWFT
shouldn't be locked.

radimkrcmar, 2025-01-02:
> First thing `kexec` flow will do is that it will take hart `satp` to `bare` mode. And that will effectively break shadow stack. If menvcfg.SSE = 1, then all shadow stack instructions will start [...]

The pre-kexec path should have no shadow stack instructions after the switch to bare mode.  (Linux uses hand-written assembly with simple jumps at that point.)
The post-kexec path must not use shadow stack before setting it up, but it is not very different from booting without shadow stacks, kernel will just skip the ecall to enable shadow stacks.

> """ If an attacker has found a vulnerability in kernel to perform a ROP, they can perform a `kexec` on a kernel which wasn't compiled with shadow stack instructions and thus completely bypassing [...]

The same goes for landing pads as a kexec kernel (any attacker provided code) will use the feature properly.

There is no need to have the lock if the control-flow protection is implemented correctly, but it is possible that the lock will force attacker to use a more sophisticated method of exploitation.  Kernel should not disable the control-flow protection on pre-kexec paths even without locking, because that provides code that the attacker could leverage.

Using the lock might not have a positive effect on security, but neither should it complicate the code of security-minded software.

deepak0414, 2025-01-02:
> The post-kexec path must not use shadow stack before setting it up, but it is not very different from booting without shadow stacks

I believe post-exec path is basically asm routine in `head.S`.
At this point shadow stack instructions can be avoided because its hand-written routine. However `setup_vm` is called from here which is a C routine and will have shadow stack instructions compiled in.
If `menvcfg.SSE=0` (first kernel boot), they will fallback to zimop.
If `menvcfg.SSE=1` due to lock installed by first kernel, they will do access/AMO store fault.

Please note that `setup_vm` is the one where early memory management stuff happens. My current RFC patches for kernel shadow stack ensures that this function sets up correct page table encodings for boot shadow stack. Shadow stack on kernel will be active only after `setup_vm` is successfully exercised.

Sure we could come with design or solutions to avoid this. But at what cost. An attacker still could bypass lock using a kernel which doesnt have shadow stack compiled in.

Thats the reason I proposed `kexec` in general goes against the philosophy of locking security feature.

>kernel will just skip the ecall to enable shadow stacks.

How? Meaning of lock means that its already enabled i.e. `menvcfg.SSE=1`. It can't be changed (it was locked to that state by previous kernel).

radimkrcmar, 2025-01-02:
> I believe post-exec path is basically asm routine in `head.S`. At this point shadow stack instructions can be avoided because its hand-written routine. However `setup_vm` is called from here which [...]

Linux can control the C compiler by marking functions with `__attribute__(())` that disables shadow stack operations.  We could easily extend it to all `__init` functions, but I think we'd have to duplicate non-init functions into `.init.text` with linking magic if setting up the page table for shadow stack reaches that far; not locking might be less work at this point.

> Sure we could come with design or solutions to avoid this. But at what cost. An attacker still could bypass lock using a kernel which doesnt have shadow stack compiled in.

My point was that the solution is not complex, even if the security benefit of locking is dubious; and supervisor software can already decide to keep shadow stack unlocked if locking doesn't make attacker's life harder than the supervisor's.

Would be nice to prove that there are no attacks that would be made simpler if the attacker could clear `menvcfg.SSE`, so we could just drop the lock without a second thought.
 
> Thats the reason I proposed `kexec` in general goes against the philosophy of locking security feature.

I do not like the locking from design perspective either -- we should be really careful about non-reversible operations.
I'd argue for removing locking for all features, but there was apparently some reason for its inclusion and it's tolerable.

> > kernel will just skip the ecall to enable shadow stacks.
> 
> How? Meaning of lock means that its already enabled i.e. `menvcfg.SSE=1`. It can't be changed (it was locked to that state by previous kernel).

Sorry, I should have worded that as "ecall that enables shadow stacks".  We can skip it if `menvcfg.SSE` is already 1.

SiFiveHolland, 2025-01-06:
> Linux can control the C compiler by marking functions with `__attribute__(())` that disables shadow stack operations. We could easily extend it to all `__init` functions, but I think we'd have to [...]

The early boot code that runs before page table initialization, including `setup_vm()`, is already isolated because it must be fully position-independent. You can search for `medany` in `arch/riscv` to see where this applies. So this should be as simple as disabling shadow stacks for those same translation units.

As a side note, the other place that requires special CFLAGS handling is calls in to EFI services. If there is no guarantee that EFI runtime services code is compiled with landing pads enabled (should that requirement be added to the UEFI spec?), then the kernel may need to disable landing pads when calling into EFI runtime services, which would also prevent that feature from being locked.

radimkrcmar, 2025-01-07:
> As a side note, the other place that requires special CFLAGS handling is calls in to EFI services. If there is no guarantee that EFI runtime services code is compiled with landing pads enabled [...]

Good point.  I think it would be best to mandate that UEFI interfaces support landing pads in the BRS spec.

atishp04, 2025-02-20:
All the comments addressed in the release https://github.com/riscv-non-isa/riscv-sbi-doc/releases/tag/v3.0-rc3

closed 2025-02-20 by atishp04